SlashDevin/NeoSWSerial #3: Doesn't compile on ATmega32U4 (Micro, Leonardo)

adienakhmad, 2016-08-07:
`error: 'TCCR2A' was not declared in this scope`
`error: 'TCCR2B' was not declared in this scope`

SlashDevin, 2016-08-09:
Confirmed.

SRGDamia1, 2017-05-13:
The 32u4 doesn't have a timer2.  Could one of the other timers be used?

SlashDevin, 2017-05-13:
Yes, but I don't have time to work on this.  Choosing the timer, clock scaler (from baud rate) and testing is what needs to be done.  If you get something working, I'd be glad to fold it in!

StackerDEV, 2018-08-27:
Somebody got it fixed?

jpmeijers, 2018-10-11:
Can you perhaps remove `ATmega32U4 (Micro, Leonardo)` from the bottom of README.md, as it is a red herring that causes a lot of wasted time. We can add it back when this issue is fixed.

(I'll file a PR)

AKopytenko, 2022-12-03:
Same problem. 
Please remove Micro from the supported boards block in README.MD