SlashDevin/NeoSWSerial #4: nss.read() not returning AT messages

shrikantbarve, 2016-09-02:
Hello,
I am using NeoSWSerial with SIM900A. I am passing few commands. When I used to do with Software Serial, I used to get what commands it threw and its response such as OK.

```
SoftwareSerial SIM900(2, 3);
Serial.println(SIM900.println("AT+SAPBR=3,1,\"Contype\",\"GPRS\""));
ShowSerialData();
Serial.println(SIM900.println("AT+SAPBR=3,1,\"APN\",\"bsnlnet\""));
ShowSerialData();
Serial.println(SIM900.println("AT+SAPBR=1,1"));
ShowSerialData();

void ShowSerialData()
{
  while(SIM900.available()!=0)
    Serial.write(SIM900.read());
}
```

But when I do this >>>

```
NeoSWSerial SIM900(2, 3);
Serial.println(SIM900.println( F("AT+SAPBR=3,1,\"Contype\",\"GPRS\"")));
ShowSerialData();
Serial.println(SIM900.println( F("AT+SAPBR=3,1,\"APN\",\"bsnlnet\"")));
ShowSerialData();
Serial.println(SIM900.println( F("AT+SAPBR=1,1")));
ShowSerialData();

void ShowSerialData()
{
  while(SIM900.available()!=0)
    Serial.write(SIM900.read());
}
```
it starts giving something like this >

```
$GPVTG,,T,,M,0.261,N,0.483,K,A*29
$GPGGA,115147.00,1906.32585,N,07250.98623,E,1,10,0.85,35.6,M,-65.6,M,,*4E
$GPGSA,A,3,29,18,15,21,20,10,13,26,25,32,,,1.56,0.85,1.30*09
$GPGSV,4,1,14,04,17,250,22,10,34,277,33,13,10,038,16,15,39,046,28*71
$GPGSV,4,2,14,16,06,294,,18,53,323,35,20,30,037,11,21,53,005,30*79
$GPGSV,4,3,14,24,19,114,23,25,06,182,15,26,11,267,22,27,03,324,*7A
$GPGSV,4,4,14,29,51,162,21,32,21,207,24*74
$GPGLL,1906.32585,N,07250.98623,E,115147.00,A,A*6F
30
$GPRMC,115148.00,A,1906.32597,N,07250.98620,E,0.321,,020916,,,A14
$GPRMC,115149.00,A,1906.32626,N,07250.98611,E,0.288,,020916,,,A14
$GPRMC,115150.00,A,1906.32622,N,07250.98610,E,0.098,,020916,,,A5,E,1,10,0.85,34.0,M,-65.6,M,,*40
$GPGSA,A,3,29,18,15,21,20,10,13,26,25,32,,,1.56,0.85,1.30*09
$GPGSV,4,1,14,04,17,250,17,10,34,277,31,13,10,038,15,15,39,046,27*79
$GPGSV,4,2,14,16,06,294,,18,53,323,35,20,30,037,19,21,53,005,33*72
$GPGSV,4,3,14,24,19,114,20,25,06,182,11,26,11,267,22,27,03,324,*7D
$GPGSV,4,4,14,29,51,162,21,32,21,207,27*77
$GPGLL,1906.32635,N,07250.98605,E,115153.00,A,A*66
13
$GPRMC,115154.00,A,1906.32632,N,07250.98604,E,0.142,,020916,,,A0.249,N,0.461,K,A*2F
$GPGGA,115157.00,1906.32639,N,07250.98602,E,1,10,0.85,33.4,M,-65.6,M,,*4C
$GPGSA,A,3,29,18,15,21,20,10,13,26,25,32,,,1.56,0.85,1.30*09
$GPGSV,4,1,14,04,17,250,18,10,34,277,32,13,10,038,13,15,39,046,27*73
$GPGSV,4,2,14,16,06,294,,18,53,323,35,20,30,037,23,21,53,005,31*79
$GPGSV,4,3,14,24,19,114,20,25,06,182,14,26,11,267,21,27,03,324,*7B
$GPGSV,4,4,14,29,51,162,18,32,21,207,27*7D
$GPGLL,1906.32639,N,07250.98602,E,115157.00,A,A*69
13

```

So I think it is a problem with NeoSWSerial read(), how can I the AT commands that arduino is passing to module ?

Thanks

SlashDevin, 2016-09-02:
That is the raw GPS data.  Because NeoSWSerial can only listen to one port at a time (like SoftwareSerial), NeoSWSerial is still listening to the GPS, not the SIM900.

For my testing, I commented out these lines in the sketch I posted:
```
//    SIM900.listen();
    SIM900INIT(message);
    SIM900STOP();
//    gps_port.listen();
```
You probably need to uncomment those two lines for your system.

shrikantbarve, 2016-09-03:
Yes, After un-commenting those lines, it has started working as expected. 

Thank you so much for your assistance and special thanks for neoswserial library.

closed 2016-09-03 by shrikantbarve