SlashDevin/NeoSWSerial #14: Unable to send sms message to a number !

AEK-BKF, 2017-07-13:
Hi man :) I hope you're fine ^^
please help me to resolve my problem !
I have Arduino Uno 3 , Sim900 GSM and GPS modules,  I used your program bellow :
```
#include <NeoSWSerial.h>
#include <NMEAGPS.h>

NMEAGPS gps;
NeoSWSerial gps_port(0, 1);

enum state_t { WAITING_TO_SEND, SENDING_CMD1, SENDING_CMD2 }; // valid FSM states
state_t  state;     // current FSM state
uint32_t stateTime; // FSM timer for delays between states

bool     newData = false; // true when a fix with a location is ready
gps_fix  fixToSend;
const uint32_t SEND_INTERVAL = 60 * 1000UL; // once a minute

void setup()
{
  Serial.begin(38400);
  gps_port.begin(38400);
}

void loop()
{
  // Always process GPS characters so the latest fix is ready to go
  while (gps.available( gps_port )) {
    gps_fix newFix = gps.read();

    if (newFix.valid.location) {
      newData   = true;
      fixToSend = newFix;
    }
  }

  // FSM for sending fixes
  switch (state) {

    case WAITING_TO_SEND:
      //  Don't send new data too often.
      if (newData && (millis() - stateTime >= SEND_INTERVAL)) {
        Serial.println( F("AT+CMGF=1") );
        stateTime = millis();
        state     = SENDING_CMD1;
      }
      break;

    case SENDING_CMD1:
      //  Delay 1 second after the CMGF command
      if (millis() - stateTime >= 1000) {
        Serial.println( F("AT+CMGS=\"+2***********\"") );
        stateTime = millis();
        state     = SENDING_CMD2;
      }
      break;

    case SENDING_CMD2:
      //  Delay 1 second after the CMGS command...
      if (millis() - stateTime >= 1000) {
        // ... then send the message
        sendFix( fixToSend );
        Serial.write(26); // no more data to send

        newData   = false;
        stateTime = millis();
        state     = WAITING_TO_SEND;
      }
      break;
  }
}

static void sendFix( const gps_fix &fix )
{
  //  Send only the fix pieces of interest
  //    (other pieces described on Data Model page)

  Serial.print( F("Latitude : ") );
  Serial.println( fix.latitude(), 6 );
  Serial.print( F("Longitude : ") );
  Serial.println( fix.longitude(), 6 );
}
``` 

I get on serial console :
```
AT+CMGF=1
AT+CMGS="+2*********"
Latitude : 36.563385
Longitude : 3.606386
AT+CMGF=1
AT+CMGS="+21*********"
Latitude : 36.563484
Longitude : 3.606415
```

But I can't receive the sms on my phone !
Thank you so much.

SlashDevin, 2017-07-13:
> NeoSWSerial gps_port(0, 1);

You shouldn't use `NeoSWSerial` on a `HardwareSerial` port.  These are the pins for `Serial`.  If your GPS device is connected to pins 0 & 1, just use `Serial`.  I *assume* that the GSM shield is also connected to pins 0 & 1, because the AT commands are also sent with `Serial`.  You can't have two devices connected to the same serial pins, unless one of them is transmit-only and the other is receive-only.

While the GPS could be used as a transmit-only device (Arduino receives), the GSM shield both receives (commands) and transmits (status).

How is everything connected?

AEK-BKF, 2017-07-13:
Thanks for quick reply :)
- Arduino uno R3
- GPS Shield 1.1 ITEAD Studio mounted on Arduino.
- GPRS/GSM Shield v1.0 mounted on GPS board.
Thanks :)

SlashDevin, 2017-07-14:
I'll ask again:

>How is everything connected?

Both of these boards have jumper blocks to select the pins to be used for the serial ports.

AEK-BKF, 2017-07-14:
I set :
SoftwareSerial serialGps(0, 1);
SoftwareSerial serialSim900(7, 8);

SlashDevin, 2017-07-14:
Pins 0 & 1 are for the `HardwareSerial` port, called `Serial`.  DO NOT use this:

```
SoftwareSerial serialGps(0, 1);   NO!
```

Instead, for the code above, use this:

```
#define gps_port Serial
```

Then use this for the GSM connection:

```
NeoSWSerial sim900( 7, 8 );
```

Also update your sketch to use `gps_port` for reading into NeoGPS (the `gps` object), and use `sime900.print` for the AT commands.

closed 2017-08-10 by SlashDevin