SlashDevin/NeoSWSerial #15: Rx or Tx only?

MrHaroldA, 2017-07-21:
It would be nice to set up NeoSWSerial to work on a single pin. For example, I use this library to fetch debug information on a Midi-enabled device which has Midi I/O attached to the hardware serial port for the least data errors.

It would be nice if I don't have to assign two ports to NeoSWSerial, as I don't need Rx.

SlashDevin, 2017-07-23:
I think I tried using an invalid pin number (e.g., 0xFF) for either pin.  Have you tried that?  I'll look at the code...

MrHaroldA, 2017-07-23:
I'll try it later this week. Still, if the library would supply an "invalid pin constant", or something like that ... that would be nice.

`NeoSWSerial ss( 4, NEOSWSERIAL_DISABLE );`

GavinAndrews, 2018-10-07:
Likewise a RX only version would be useful

caradag, 2020-09-20:
Any updates on this?
Looking for RX only. Would an invalid pin work with no side effects?
If I give it a pin that I already use as OUTPUT for something else but never transmit over that serial connection. Would the serial connection interfere with whatever I'm doing in that pin? Or viceversa?

nicdu40, 2021-02-08:
hello , i modify library , and it works for RX only : NeoSWSer MySerial ( RX_PIN );
[NeoSWSerial_RX_only.zip](https://github.com/SlashDevin/NeoSWSerial/files/5947528/NeoSWSerial_RX_only.zip)