SlashDevin/NeoSWSerial #16: Restructure interrupts to allow an external interrupt routine to call by pin

SRGDamia1, 2017-09-11:
To allow multiple types interrupts on the same vector

SlashDevin, 2017-09-12:
[This pull request](https://github.com/SlashDevin/NeoSWSerial/pull/10) added this capability.  I see that the Readme does not describe the capability, so I need to update it.  There's a good summary on the Arduino forum, [here](https://forum.arduino.cc/index.php?topic=494806.msg3377416#msg3377416).

SRGDamia1, 2017-09-12:
Thank you!  I guess I'm a little confused by how this would work.  If I call the interrupt with Grey Gnome's Enable Interrupt, will NeoSWSerial take over the whole interrupt "vector" (ie, all 8 pins) or will it only grab the one Rx pin?

SlashDevin, 2017-09-13:
[EnableInterrupt](https://github.com/GreyGnome/EnableInterrupt) is in charge of the ISRs (i.e., vector for the whole port) and dispatching to the registered functions.

`NeoSWSerial::rxISR` is simply one of the registered functions.  NeoSWSerial will not define any ISRs when that define is uncommented.  It's up to you (or one of your libraries) to call `NeoSWSerial::rxISR` when you know the pin has changed.  You could simply poll the pin state, but that makes it susceptible to processing delays.  However, this could cause NeoSWSerial to generate incorrect bits because the timings are off.

`AltSoftSerial` is better because it uses the Input Capture capabilites of the MCU.  The MCU will save a timestamp of when the bit changed.  This allows a little more flexibility in servicing the ISR.  Even if the ISR is not called immediately, the timestamp will let `AltSoftSerial` know *when* the bit changed, not just that it changed *sometime* in the past.

SRGDamia1, 2017-09-13:
Ok, got it.

I know about AltSoftSerial and agree that it's more stable/accurate (and also doesn't have any ISR conflicts).  But I also like that this gives the option of (almost) any pin, and it still seems a bit better than the standard SoftwareSerial.

closed 2017-09-13 by SRGDamia1