SlashDevin/NeoSWSerial #21: Doesn't compile on ATTiny84

gion86, 2017-10-01:
I have the same error as in issue #3.
Now I could try to use a different timer on the tiny84, timer1 should be ok??
The problem is that, from the datasheet, I see the prescaler can only be: 1, 8, 64, 256.... **but no 32**.
So is there anything else I have to change to make it work on the tiny84 with the prescaler at 64 for example??
Should be something similar to the work of @SRGDamia1 for AtMega32...

Thanks!

SRGDamia1, 2017-10-03:
On the 32u4, I solved the missing timer 2 by picking another timer (#4) with the right prescaler.  It doesn't look like the ATtinyx4 has that option.  I think with a different prescaler/timer speed you'd need to adjust the TICKS_PER_BIT and BITS_PER_TICK values for that speed (similar to the discussion in #19)

gion86, 2017-10-04:
**Grazie Sara! ;)**

I've made the changes to compile and run on the Tiny84 (I will eventually create a pull request):
`TCCR1A = 0x00; // "normal" operation - Normal port operation, OC1A & OC1B disconnected`
   `   TCCR1B = 0x03; // 0b00000011 - Clock Select bits 11 & 10 on - prescaling source = CK/64`

The timer is timer 1, and PCI_FLAG_REGISTER is GIFR (same as Tiny85).

Now that the timer is running at 8MHz/64 = 125kHz:

`static const uint8_t  TICKS_PER_BIT_9600      = 13;     // 9600 baud bit width in units of 8us`
`static const uint8_t  TICKS_PER_BIT_31250     = 4;      // 31250 baud bit width in units of 8us`
`static const uint16_t BITS_PER_TICK_31250_Q10 = 256;    // 31250 bps * 0.000008 s * 2^10 "multiplier"`
`static const uint16_t BITS_PER_TICK_38400_Q10 = 314;    // 1s/(38400 bits) * (1 tick)/(8 us) * 2^10  "multiplier"`