SlashDevin/NeoSWSerial #31: NeoSWSerial with PinChangeInterrupt

geologic, 2019-02-01:
Hi

I'm using your library to talk to a GSM modem:

`NeoSWSerial modemSerial (GSM_TXPIN,GSM_RXPIN);        // Pins 7,8 `

I also use avr/wdt.h for watchdog functions, and 2 external interrupts (pin d2 and d3). Everything is working ok.

I need to check a 3rd pin connected to a pushbutton, so i made a new sketch just with PinChangeInterrupt library where i check for a button press on XTAL2 pin (i don't use xtal):

`attachPCINT( 7, gotPressP1, FALLING);      // Check PCINT=7 -> PB7 = XTAL2`

 That also works ok.
When i try to merge everything, i got linking errors, so i try to disable NeoSWSerial ISRs: on .h file i uncomented 

`#define NEOSWSERIAL_EXTERNAL_PCINT // uncomment to use your own PCINT ISRs`

and added ISR to manage RX:

```
attachPCINT(7, gotPressP1, FALLING);
attachPinChangeInterrupt(digitalPinToPinChangeInterrupt(GSM_RXPIN), ISRrx, CHANGE);

void ISRrx() {
  NeoSWSerial::rxISR( *portInputRegister( digitalPinToPort( GSM_RXPIN ) ) );
}
```

Now it compiles ok but my board is always reeboting, and i can see it is a watchdog reset.

Any idea why?

SRGDamia1, 2019-02-01:
The problem is with using the XTAL pin as input, nothing caused by this library.  I strongly suspect that you have to do some tweaking to pins_arduino.h before that would work.

Otherwise this library works very nicely when you attach the pin change interrupts with some other library, just like you're already doing.

geologic, 2019-02-04:
Well, i'm prety sure the problem is not XTAL pin as input, as i have a standalone working sketch using that pin and PinChangeInterrupt library.

I made onother test: I grab my original GSM code (working)  where i use NeoSWSerial and just change it to disable  NeoSWSerial ISRs, like stated on point 6)
Code compiles ok but modem does not respond correctly anymore.
So, definitely is something between NeoSWSerial and PinChangeInterrupt libraries. 

If i try to use EnableInterrupt instead of PinChangeInterrupt i got compile errors: multiple definition of `__vector_1' and multiple definition of `__vector_2'

SRGDamia1, 2019-02-04:
I've never used PinChangeInterrupt, but I have NeoSWSerial working well with EnableInterrupt.  EnableInterrupt won't use the XTAL pin though.  Do you really need that pin?  Have you tried with another?

All I did to get it working with EnableInterrupt:
Un-comment the line: (or compile with build flag -DNEOSWSERIAL_EXTERNAL_PCINT)
```#define NEOSWSERIAL_EXTERNAL_PCINT // uncomment to use your own PCINT ISRs```

Created an ISR helper fxn:
```cpp
void neoSSerial1ISR()
{
    NeoSWSerial::rxISR(*portInputRegister(digitalPinToPort(neoSSerial1Rx)));
}
```

And called the ISR in my setup fxn:
```enableInterrupt(neoSSerial1Rx, neoSSerial1ISR, CHANGE);```

geologic, 2019-02-05:
After trial and error i manage to put everything together... 
My original code works ok with NeoSWSerial and a GSM modem.
Adding PinChangeInterrupt works ok and my pin responds to events, but GSM no longer work. 
The problem seems to be in ISR registration, since GSM no longer receive or send AT commands.

So, i made another try: used my original code with EnableInterrupt library (instead of PinChangeInterrupt), and done the suggested changes to ISR function, but code does not compile: 
`multiple definition of '__vector_1' and multiple definition of '__vector_2'`

albi90, 2020-02-02:
> After trial and error i manage to put everything together...
> My original code works ok with NeoSWSerial and a GSM modem.
> Adding PinChangeInterrupt works ok and my pin responds to events, but GSM no [...]

Im also seeing a similar issue with a Lora Chip & NeoSWSerial + a pin change interrupt the interrupt fires then i loose connection to my Lora chip (using pins 8 and 9 so not XTAL) did you work out the issue?

geologic, 2020-02-05:
> 
> Im also seeing a similar issue with a Lora Chip & NeoSWSerial + a pin change interrupt the interrupt fires then i loose connection to my Lora chip (using pins 8 and 9 so not XTAL) did you work out the issue?

No, i give up using NeoSWSerial and changed my 328P for the 328PB version, since 328PB has 2 UARTS. 
Careful, 328P and 328PB are not pin compatible, you have to check your design.