SlashDevin/NeoSWSerial #37: NeoSWSerial not working with Bluetooth Bee Standalone module

TedJackson, 2020-01-02:
Hi.  I need to read from two serial streams at 9600 and 38400 baud with my Bluetooth Bee Standalone (from SeeedStudio, I believe).  NeoSWSerial seemed just the thing.  However I'm getting:

#error MCU not supported by NeoSWSerial!
and
Error compiling for board Arduino BT.

Was wondering if there might be a work-around, if I'm possibly doing something wrong, or if support might be extended to its MCU.  Anyone know how to find out what it's MCU is exactly?  I think it's an ATMega168.  Thanks.

BlaT2512, 2020-01-16:
Hi TedJackson,
You are right, the Bluetooth Bee Standalone has an ATmega168 chip. The `#error MCU not supported by NeoSWSerial!` means that NeoSWSerial doesn't support your device/chip. You can see in this piece of code here from NeoSWSerial.cpp that defines vectors that it returns the error you received because your chip didn't match any of the if statements:

```
#if defined(__AVR_ATtiny261__) | \
      defined(__AVR_ATtiny461__) | \
      defined(__AVR_ATtiny861__)

  ISR(PCINT0_vect)
  {
    if (GIFR & _BV(INTF0)) {
      NeoSWSerial::rxISR(PINA);
    } else {
      NeoSWSerial::rxISR(PINB);
    }
  }

  #elif defined(__AVR_ATtiny25__) | \
        defined(__AVR_ATtiny45__) | \
        defined(__AVR_ATtiny85__) 

  PCINT_ISR(0, PINB);

  #elif defined(__AVR_ATtiny24__) | \
        defined(__AVR_ATtiny44__) | \
        defined(__AVR_ATtiny84__) 

  PCINT_ISR(0, PINA);
  PCINT_ISR(1, PINB);

  #elif defined(__AVR_ATmega328P__)

  //PCINT_ISR(0, PINB);
  PCINT_ISR(1, PINC);
  //PCINT_ISR(2, PIND);

  #elif defined(__AVR_ATmega32U4__)

  PCINT_ISR(0, PINB);

  #elif defined(__AVR_AT90USB1286__)

  PCINT_ISR(0, PINB);

  #elif defined(__AVR_ATmega2560__)

  PCINT_ISR(0, PINB);
  PCINT_ISR(1, PINJ);
  PCINT_ISR(2, PINK);

  #elif defined(__AVR_ATmega1281__)

  PCINT_ISR(0, PINB);
  // PCINT8 on PE0 not supported.  Other 7 are on PJ0..6
  PCINT_ISR(1, PINJ);
  PCINT_ISR(2, PINK);

  #elif defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644P__)

  PCINT_ISR(0, PINA);
  PCINT_ISR(1, PINB);
  PCINT_ISR(2, PINC);
  PCINT_ISR(3, PIND);

  #elif defined(__AVR_ATmega2560RFR2__)

  PCINT_ISR(0, PINB);
  PCINT_ISR(1, PINE);

  #else
    #error MCU not supported by NeoSWSerial!
  #endif
```

You said that you need to read from two serial streams on your device. What are the two streams from? Is one of the built in Bluetooth on the device? I may be able to help you write vector code for this device if you can provide info for what you need to get serial data from.

Regards,
Blake.

TedJackson, 2020-01-18:
Hi Blake,One stream is the Bluetooth (read and write) as you correctly guessed.  The other simply reads data from an RFID module at only 9600 baud.  I can do the second on any pin, if only a certain pin on the BT Standalone can be supported by NeoSWSerial.Thanks much,  Ted-----Original Message-----
From: Blake Tourneur 
Sent: Jan 16, 2020 6:18 PM
To: SlashDevin/NeoSWSerial 
Cc: TedJackson , Author 
Subject: Re: [SlashDevin/NeoSWSerial] NeoSWSerial not working with Bluetooth Bee Standalone module (#37)

Hi TedJackson,
You are right, the Bluetooth Bee Standalone has an ATmega168 chip. The #error MCU not supported by NeoSWSerial! means that NeoSWSerial doesn't support your device/chip. You can see in this piece of code here from NeoSWSerial.cpp that defines vectors that it returns the error you received because your chip didn't match any of the if statements:
#if defined(__AVR_ATtiny261__) | \
      defined(__AVR_ATtiny461__) | \
      defined(__AVR_ATtiny861__)

  ISR(PCINT0_vect)
  {
    if (GIFR & _BV(INTF0)) {
      NeoSWSerial::rxISR(PINA);
    } else {
      NeoSWSerial::rxISR(PINB);
    }
  }

  #elif defined(__AVR_ATtiny25__) | \
        defined(__AVR_ATtiny45__) | \
        defined(__AVR_ATtiny85__) 

  PCINT_ISR(0, PINB);

  #elif defined(__AVR_ATtiny24__) | \
        defined(__AVR_ATtiny44__) | \
        defined(__AVR_ATtiny84__) 

  PCINT_ISR(0, PINA);
  PCINT_ISR(1, PINB);

  #elif defined(__AVR_ATmega328P__)

  //PCINT_ISR(0, PINB);
  PCINT_ISR(1, PINC);
  //PCINT_ISR(2, PIND);

  #elif defined(__AVR_ATmega32U4__)

  PCINT_ISR(0, PINB);

  #elif defined(__AVR_AT90USB1286__)

  PCINT_ISR(0, PINB);

  #elif defined(__AVR_ATmega2560__)

  PCINT_ISR(0, PINB);
  PCINT_ISR(1, PINJ);
  PCINT_ISR(2, PINK);

  #elif defined(__AVR_ATmega1281__)

  PCINT_ISR(0, PINB);
  // PCINT8 on PE0 not supported.  Other 7 are on PJ0..6
  PCINT_ISR(1, PINJ);
  PCINT_ISR(2, PINK);

  #elif defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644P__)

  PCINT_ISR(0, PINA);
  PCINT_ISR(1, PINB);
  PCINT_ISR(2, PINC);
  PCINT_ISR(3, PIND);

  #elif defined(__AVR_ATmega2560RFR2__)

  PCINT_ISR(0, PINB);
  PCINT_ISR(1, PINE);

  #else
    #error MCU not supported by NeoSWSerial!
  #endif

You said that you need to read from two serial streams on your device. What are the two streams from? Is one of the built in Bluetooth on the device? I may be able to help you write vector code for this device if you can provide info for what you need to get serial data from.
Regards,
Blake.

BlaT2512, 2020-01-18:
Hi Ted,
Interesting that you have a serial RFID card reader. I use the common RC522 card reader, which uses SPI. For programming the device, do you use an ISP device (such as USBISP or AVR ISP mkII) or the official programming device (Xbee carrier / UartsSBee)?

Depending on what serial library you use for connecting to the bluetooth device on the board, it may support multiple instances. Please send your code and I will write something for you to work for both devices. 

Thanks,
Blake.

TedJackson, 2020-01-18:
Hi Blake,Thank you for your interest and attention.  Actually, it's a Chinese board for long distance (~15cm) tag detection (ISO11784/ISP11785) with only serial out capability.  I haven't been able to find anything with that long a range elsewhere (and I need that range).  I'm only just now getting deeper into PCB design and manufacture (and RFID transponder design) and graduating from the grade school of Arduino Bee simplicity to the college level of STM8 MCU complexity/versatility.  I see that ST has some RFID offerings, which I'll look into.  I've been using the standard Serial (for debugging with a terminal) and two instances of SoftwareSerial for the BT and RFID, till I realized that the 'only one one listener at a time' was causing some of the problems I was experiencing at which point I started looking for more options like NewSWSerial and AltSoftSerial.  My code is in development, so there's nothing permanent to send you.  But I thank you for the offer.Thanks and regards,  Ted-----Original Message-----
From: Blake Tourneur 
Sent: Jan 18, 2020 4:16 AM
To: SlashDevin/NeoSWSerial 
Cc: TedJackson , Author 
Subject: Re: [SlashDevin/NeoSWSerial] NeoSWSerial not working with Bluetooth Bee Standalone module (#37)

Hi Ted,
Interesting that you have a serial RFID card reader. I use the common RC522 card reader, which uses SPI. For programming the device, do you use an ISP device (such as USBISP or AVR ISP mkII) or the official programming device (Xbee carrier / UartsSBee)?
Depending on what serial library you use for connecting to the bluetooth device on the board, it may support multiple instances. Please send your code and I will write something for you to work for both devices.
Thanks,
Blake.

BlaT2512, 2020-01-19:
Hi Ted,
Thanks for the information.
If you still need help with this, open the NeoSWSerial.cpp file (probably in <Arduino Sketchbook Location>/libraries/NeoSWSerial/src) and add in vector code (in my first reply) after
```
#elif defined(__AVR_ATmega2560RFR2__)

  PCINT_ISR(0, PINB);
  PCINT_ISR(1, PINE);
```
and before 
```
#else
    #error MCU not supported by NeoSWSerial!
 #endif
```
So the end should look like
```
#elif defined(__AVR_ATmega2560RFR2__)

  PCINT_ISR(0, PINB);
  PCINT_ISR(1, PINE);

#elif defined(__AVR_ATmega168__) || defined(__AVR_ATmega168P__)

  PCINT_ISR(0, PIN0);
  PCINT_ISR(1, PIN1);
  PCINT_ISR(2, PIN2);

#else
    #error MCU not supported by NeoSWSerial!
#endif
```