SlashDevin/NeoSWSerial #45: Nano RP2040 Connect board support

santanig5, 2021-05-26:
Greetings,
I tried to upload a sketch I've used with the Arduino Nano, ATmega328P to the new Nano RP2040 Connect board and received many errors.
Do you plan for your library to support this board? If so do you have a time frame?
Kind Regards
[Nano RP2040 Errors.txt](https://github.com/SlashDevin/NeoSWSerial/files/6550056/Nano.RP2040.Errors.txt)

SRGDamia1, 2021-05-27:
I'm not the maintainer on this, but it look from your log like NeoSWSerial isn't the cause of most of your errors.

santanig5, 2021-05-27:
Hi,
Maybe not most, but some.
When I sent the logs to Arduino support they identified
NeoSWSerial as one of the offending libraries.
I've opened a similar case with the other.
Thanks

Sent from my Verizon, Samsung Galaxy smartphone